cano64/ArduinoSystemStatus #5: ATTiny85V slightly inaccurate voltage reading

Andon-A, 2017-05-27:
Having a slight issue on my ATTiny85v.

Using SystemStatus to grab the VCC is great, and super handy. It is, however, wrong.

I have an Arduino Uno providing power, via the 5v pin. My voltmeter registers it at roughly 4.82v, and using SystemStatus on the Uno returns about 4830-4850 mV most of the time. Slightly inaccurate, but very, very close.

The ATTiny85V (Using TinyDebugSerial and TinyWireS) is reporting 5200-5300 mV. This is clearly significantly off. Additionally, if I switch it to the 3.3v output, it reads at around 3420-3440mV. My voltmeter registers it at 3.31v, so again, it's off by a decent margin.

For complete information, I have both the Uno and the 85V running the code side-by-side, measuring the same voltage. Uno just spit out 4834, the ATTiny 5313 - Almost a 10% difference.

Granted, these are different enough that I can still utilize them, but they're off by a significant amount. I know the ATTiny85V has a wider operating voltage range than the standard ATTiny85 (That's why I chose it), and I THINK it has different internal registers, but I'm not sure on that.

Andon-A, 2017-05-28:
As an update, it seems that it's slowly going closer to where it should be. I'm thinking there might be a burn-in time, as it was a brand new chip.

cano64, 2017-05-30:
The bandgap reference voltage could be as 10% off and varies from chip to chip. It should be stable thought, so if you calibrate it you should get accurate reading. https://en.wikipedia.org/wiki/Bandgap_voltage_reference

closed 2017-05-30 by cano64